aredn/aredn #2367: You cant set the GPS coordinates on a node if the ARDEN web certificate is down

K1RKS, 2025-07-31:
A newly created node, updated to the latest nightly: 20250723-c45cd17e

Was asking for the GPS coordinates to be set.

 I tried to set it multiple times (even turning off the advanced setting:

GPS Location
Use local or network GPS to set location

And after you hit "DONE" it clears back to not set.

nc8q, 2025-07-31:
I do not understand.
Can you set the longitude, latitude, and grid square on the node
if it us unplugged from the internet? 
I can.
The world map is available to me.
Is there really a Global Positioning System device somewhere or 
are you just trying to set the longitude, latitude, and grid square?

73, Chuck

K1RKS, 2025-07-31:
I guess I could try unplugging it from the internet to try.
All I know is if I manually put the GPS co-ordinates in (like I've done 100 times so far) .. hit "Done" .. it comes back w/o GPS coordinates.
Steps:
1)  Here is how it starts.

[image]

2) Set the coordinates:

[image]

3) Hit Done

It goes to here. (One time it said "commit" and I hit commit but it still came back to here)

[image]

aanon4, 2025-09-30:
Fixed?