aredn/aredn #2388: Gl-inet release 3.25.8.0 looses internet via a RF connected device

W0zxh, 2025-08-19:
### Describe the bug
A clear and concise description of what the bug is.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] yes #.25.8.0

### Expected behavior
A clear and concise description of what you expected to happen.
Expected no change from version 24

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

Clients PC Any node connected via RF to main connected internet mesh unit will not let clients access Internet. All AREDN routers however can ping and trace route just fine. Clients had Internet access on previous version and after regressing back the work again.

ab7pa, 2025-08-19:
https://docs.arednmesh.org/en/latest/arednGettingStarted/node_admin.html#support-data

aanon4, 2025-08-20:
As this node doesnt have Internet itself, could you attach the support data for the node which does. Thanks.

W0zxh, 2025-08-20:
pc always works
                                                              |
pc--->first router GLinet --> tunnel glinet-->internet 
no work             works             always  works               3.28.8.0
works                works              always works
                                                         3.25.8.0

[supportdata-W0ZXH-TW1-2025-8-20-19-16.tar.gz](https://github.com/user-attachments/files/21903964/supportdata-W0ZXH-TW1-2025-8-20-19-16.tar.gz)

W0zxh, 2025-08-21:
GLinet were not
Ubiquiti was
Funny then as it worked on last version

However had to go will test tomorrow 
Thanks for your assistance W0ZXH JERRY

aanon4, 2025-08-21:
It was broken on the last version.

W0zxh, 2025-08-21:
support data

W0zxh, 2025-08-21:
re-enabling mesh to wan mesh to remote an fixed rf only routers