aredn/aredn #2415: Still having routing issues since leaving 3.25.5.1

edkraus, 2025-09-03:
### Describe the bug
I am having ongoing routing issues where nodes still appear in the node list, either Babel or OLSR but cannot be reached.  These are nodes that were rock solid a few weeks ago.  There is a path I can follow from my location to the nodes and associated services there and I can only hop from one node to another until I reach a particular SXT5ac and cannot go further to the next step a Rocket ac Lite with Omni.  The rf path is only a few hundred feet and rock solid so something is going on with the routing.  Once I reboot the SXT5 then the path becomes usable again ... for a while.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes.  Every node in this path is on the 8/30 nightly.

### Expected behavior
I want nodes to be reachable when I click on them, as well as the services attached to their LAN.  I'd like not to have to keep rebooting nodes to be able to get around the network.

### Support data
You will find several versions attached of support data for node (C) which is the SXT5ac.  I also am attaching some from the node (D) the Rocket ac lite.  The most recent for C is while unable to get to node D.  If you need a reboot and an immediate redo of the support data while it's working I can do that.

### Screenshots
I don't know what screenshots would be relevant.  

### Additional context
This is the same network and sites described earlier on the WVMN forum, in case you have been following it.  Different date support data for the same node might follow sequential attempts to go from 3.25.8.0 thru the 8/30 nightly.

[D-supportdata-K0HSU-BRB-OMNI-2025-8-25-8-1.tar.gz](https://github.com/user-attachments/files/22106048/D-supportdata-K0HSU-BRB-OMNI-2025-8-25-8-1.tar.gz)
[D-supportdata-K0HSU-BRB-OMNI-2025-8-27-6-54.tar.gz](https://github.com/user-attachments/files/22106046/D-supportdata-K0HSU-BRB-OMNI-2025-8-27-6-54.tar.gz)
[D-supportdata-K0HSU-BRB-OMNI-2025-8-31-11-26.tar.gz](https://github.com/user-attachments/files/22106047/D-supportdata-K0HSU-BRB-OMNI-2025-8-31-11-26.tar.gz)

[C-supportdata-KE7AWR-OHSU-SXT3-2025-8-25-8-6.tar.gz](https://github.com/user-attachments/files/22106033/C-supportdata-KE7AWR-OHSU-SXT3-2025-8-25-8-6.tar.gz)
[C-supportdata-KE7AWR-OHSU-SXT3-2025-8-27-6-51.tar.gz](https://github.com/user-attachments/files/22106031/C-supportdata-KE7AWR-OHSU-SXT3-2025-8-27-6-51.tar.gz)
[C-supportdata-KE7AWR-OHSU-SXT3-2025-8-31-11-4.tar.gz](https://github.com/user-attachments/files/22106032/C-supportdata-KE7AWR-OHSU-SXT3-2025-8-31-11-4.tar.gz)
[C-supportdata-KE7AWR-OHSU-SXT3-2025-9-2-17-24.tar.gz](https://github.com/user-attachments/files/22106034/C-supportdata-KE7AWR-OHSU-SXT3-2025-9-2-17-24.tar.gz)

aanon4, 2025-09-03:
And I should mention that once this happens, rebooting wont help you, although a power cycle of the relevant nodes probably will.

edkraus, 2025-09-03:
> Congratulations! You are experiencing this irritating little Babel bug where the radios refuses to sync with each other. Currently we dont know why this happens and are unable to reproduce it by hand which makes fixing it tricky/impossible.
> We do have a work around in the code, but it's extremely conservative as to when it gets triggered. It currently requires a perfect connection and, according to the support data, the link between these two nodes is not quite that - which is fine, but means the mitigation wont fire.
> So ... I'm going to make it less conservative tonight, then tomorrow if you could try to nightly build I would appreciate it.
> Thanks.

Oh boy ... 
Q ... have you read my post at WVMN?  There are five nodes involved, one tunnel, two dtd at two different remote sites with controlled access.  I might be able to power cycle the SXT but I cannot remotely power cycle the Rocket.  
One site has seven nodes dtd on two switches.  It's not a simple thing to mess with this.
I've got another nearby site with five nodes that is a secured facility it takes me about three months to get permission to access.  I can get LAN access via wifi thru the fence ... possibly if I have to reboot something there but I'm really scared to update any of those nodes which are all 3.25.5.1 for now.
So ... I need to measure three or four times and hopefully only cut twice.  Tell me what specific information (including if I need to post back here the thread at WVMN).
Thanks again,
Ed Krausf.t.n.m.p.

edkraus, 2025-09-03:
OK, I uploaded the Sept 3 nightly to the Rocket about ten minutes ago.  It's back showing up and I can get to it, but I cannot reach now the hAP lite and it's service/server on the hAPs LAN.  The hAP shows up on the OLSR routing search, but not the Babel routing search.  I can see the link from the Rocket but clicking it goes nowhere.

I'll wait then send the support data but now I can't reach the server.  So if the Omni has freshly rebooted that concerns me.

edkraus, 2025-09-03:
Now I cannot reach the Rocket or the hAP via any route.  Weird.

KG7GDB, 2025-09-03:
Thanks Ed for reporting this. We have multiple sites in WVMN Oregon which are experiencing the same unexpected loss of node access after updating to 3.25.8.0. Usually the update to nightly builds 2025-0825 or later helped with access. I was assuming the DTD RTT was the only issue. However if the cause included a stringent 100% LQ requirement for local nodes, this would still result in problems. It is very difficult to maintain 100% LQ on any mesh node. 
So I hope #2416 resolves this. For mesh, we often need to be more forgiving of poorer LQ. Under the old LQM feature, I would sometimes reduce the LQ requirement down to 30% to maintain a marginal but needed RF connection.

aanon4, 2025-09-03:
@KG7GDB Best not conflate the general use of LQ which what's being discussed here - they are not related I'm afraid.

aanon4, 2025-09-03:
@edkraus Looks like the recovery code still did not trigger. At this point I'm going to need access to debug this - sorry. Would that be possible? I'd need to get into both the nodes originally reported the SXT3 and the OMNI. Thanks.

edkraus, 2025-09-03:
I'll send you via separate email a password for the Omni.  The same owner is rf on both sides to the Omni so same issue.  The Omni has the hAP dtd.

ab7pa, 2025-09-03:
Babel restart: https://docs.arednmesh.org/en/latest/arednGettingStarted/node_admin.html#advanced-internal-service-options

edkraus, 2025-09-03:
Yep.

Related question/observation ... 

In the past if I made a change to a remote node (adding local service for example) it would take some time for the changes to propagate node to node via OLSR so I would be able to navigate from my location via hostname.  Ok, I would get around that by using my knowledge of the network topology to go directly to a neighbor node, then access the service from there as the neighbor would be the first one to the new routing info.  Since 3.25.5.1 this no longer works ... and it takes a long time for me to access anything I've changed in DHCP remotely.  I'm guessing that the change is not propagating thru the network as quickly as it used to?

aanon4, 2025-09-03:
Let's not muddy this issue with other things. The quick answer is "service resolution will be better once we're babel-only and messy until then"

edkraus, 2025-09-03:
Now traceroute takes me via an rf path of marginal quality I forgot even exists.  KK7GFI nodes are on the side of a house, one goes to OHSU another to Terwilliger.  I haven't seen routing though him for a long while.

aanon4, 2025-09-03:
I cannot speak to that. This bug is about one very specific issue on one clearly identified link.

edkraus, 2025-09-04:
Tim, the latest version you uploaded made it through the night.  All the nodes you patched are up and reachable as well as their associated serviced.  I believe you said the patches are in the Sept 4 nightly so it might be worth more people testing.

Any changes I'll post back here but you might have killed this bug (fingers crossed)

73   Ed

aanon4, 2025-09-09:
@k1ky Happy to handle it here. But without more information there's nothing I can do currently.

aanon4, 2025-09-09:
So .. this is two sides of a working connection? That sounds good.

k1ky, 2025-09-09:
Here is another dead RF Babel node

[supportdata-K1KY-LHG5-152-SMYRNA-OFFICE-COUCH-2025-9-9-10-23.tar.gz](https://github.com/user-attachments/files/22237665/supportdata-K1KY-LHG5-152-SMYRNA-OFFICE-COUCH-2025-9-9-10-23.tar.gz)

edkraus, 2025-09-09:
> I just rolled a new babel build (two in a day!) because some of the timeout were too short and casusing Babel to shutdown under some circumstances. Give it a try.

So no system data needed for now?
Which node should this reside on, the last one I could contact or the one I lost contact with?
Ed Krausf.t.n.m.p.

aanon4, 2025-09-09:
Thanks.
Let's get this stuff on today's build, including the SXT3 heading into them, and go from there.

edkraus, 2025-09-09:
OK, I started with the HAP which is furthest from my location.  I've attempted to update twice but the process will not complete, either the upload line left to right stalls, or the back and forth never goes to the updating screen.  I have to exit my browser and restart to access the node I'm currently on the LAN of.  When I finally get back to the HAP it is still on 0904 but this time I can't get back to the HAP at all.
I could reboot the OMNI, I could attempt to update the OMNI.  Which process do you prefer?
Ed Krausf.t.n.m.p.

edkraus, 2025-09-09:
>  attached is support data for all three nodes, as of now, before any updates
> Ed Krausf.t.n.m.p.

I could not get to the HAP so I started with the SXT, then the OMNI and did the update.  
Now, the OMNI reports to the HAP 100/100 lq/nlq and status excellent.  I cannot reach the HAP at all.  I cannot reach it nor any of the services.  It's still on 0904.  If this persists I'll have to make a site visit.  OMNI support data as of now attached.
Ed Krausf.t.n.m.p.

aanon4, 2025-09-09:
I can reach K0HSU-BRB-HAP - is that the one you cannot get to?

aanon4, 2025-09-10:
If you can get to the HAP to change the password to the one you shared with me once, I can probably update it by hand

edkraus, 2025-09-10:
I cannot get to the HAP using either OLSR or Babel.  I'm going to have to schedule an onsite visit and an update from the LAN.
Can you verify for me that 0909 has been used ok on these type nodes already?  I'm getting really nervous about more changes.
Ed Krausf.t.n.m.p.

edkraus, 2025-09-10:
However, right now I can get into the services on the attached ProxMox server.  Weird.  I guess I could ssh into the HAP if I can remember the ip address (of course I can find it).  Are you going to suggest I upload the file via scp into a tmp directory somewhere and do a linux install from cli?
Ed Krausf.t.n.m.p.

edkraus, 2025-09-10:
weird again, all of a sudden while looking for the ip address I got into the node via the gui.  I've turned off the wifi to see if that frees up memory and it's rebooting.  I'll attempt again to apply 0909
Ed Krausf.t.n.m.p.

edkraus, 2025-09-10:
OK.  All three nodes are on 0909.... support data attached.

[supportdata-K0HSU-BRB-HAP-2025-9-9-19-31.tar.gz](https://github.com/user-attachments/files/22246157/supportdata-K0HSU-BRB-HAP-2025-9-9-19-31.tar.gz)
[supportdata-K0HSU-BRB-OMNI-2025-9-9-19-32.tar.gz](https://github.com/user-attachments/files/22246156/supportdata-K0HSU-BRB-OMNI-2025-9-9-19-32.tar.gz)
[supportdata-KE7AWR-OHSU-SXT3-2025-9-9-19-33.tar.gz](https://github.com/user-attachments/files/22246155/supportdata-KE7AWR-OHSU-SXT3-2025-9-9-19-33.tar.gz)

edkraus, 2025-09-10:
Ooops.  I posted the support data on Github directly and by mistake closed the issue.  Not what I intended to do
Ed Krausf.t.n.m.p.