aredn/aredn #2421: HaLow - MorseMicro EKH03-03 not connecting to WAN

darrylhodgins, 2025-09-03:
### Describe the bug

Unable to get a WAN IP regardless of whether using untagged port or tagged VLAN.

- [✅] Does this issue still occur on the latest version? (Yes; tested on `babel-20250903-b806fbe0`)

### Expected behavior

System is able to connect to a WAN uplink

### Support data

[supportdata-VE7ZDH-halow-1-2025-9-3-15-25.tar.gz](https://github.com/user-attachments/files/22128034/supportdata-VE7ZDH-halow-1-2025-9-3-15-25.tar.gz)

### Screenshots

#### Network status on node:

[image]

#### Network config on node:

[image]

#### Port config

[image]

#### Status

[image]

### Additional context

* DtD (on VLAN `2`) works.
* I have another node (`VE7ZDH-wrt-1`) which has a working WAN connection.

aanon4, 2025-09-03:
The internal network configuration looks okay.
What specific device are you using?
When you say you have another node working, is this identical and assuming not how is that different this configuration?

darrylhodgins, 2025-09-03:
I'm using the Heltec [HT-HD01 HaLow Dongle](https://heltec.org/project/ht-hd01/).

My other node that has a WAN connection is a Banana Pi OpenWrt One MediaTek MT7981B, which has a very different configuration 😃

darrylhodgins, 2025-09-04:
Found the problem: one of my switches was blocking the VLAN.  Nothing wrong with the node.

closed 2025-09-04 by darrylhodgins