aredn/aredn #2427: Babel and Camera Operation vs 3.25.2.0

denis133, 2025-09-05:
###I updated all my nodes to the latest firmware version, 3.25.8.0, and immediately noticed that our fire tower cameras became extremely glitchey, running slowly, freezing, and generally misbehaving. After troubleshooting, I reverted all network nodes back to 3.25.2.0, and everything returned to normal. Honestly, 3.25.2.0 has been the most stable and reliable firmware I’ve ever used, zero issues. So what’s going on with all Babel firmware?
All the nodes except one are relatively new MikroTik devices with plenty of memory. The only older unit is an Ubiquiti PowerBridge. Here’s something else that threw me for a loop: I updated two nodes that were running on channel 142, and one of them located on the fire tower randomly switched to channel 175 after the firmware upgrade. It drove me nuts trying to figure out why it didn’t come back online after updating from 3.25.2.0 to 3.25.8.0. A WiFi scan showed no other node on channel 142 but there was one on 175, and I did not have any nodes on 175 but sure enough, that’s where it had landed. 
I later found another node in one of our arrays that had done the same thing landed on 172. I also don’t have anything running 172.  It wasn’t connected to anything, so it slipped past me initially.
Please find the Support information for the issue and also I found that the PiAware would not send updates to Flight Aware on last night’s Nightly Build. Everything is now again back to 3.25.2.0 all is good. I don’t want to be a holdout and nor move to the current builds but it just doesn’t work here.
 Describe the bug
A clear and concise description of what the bug is.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Operate as it does on 3.25.2.0
### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-WA1USA-CHOPMIST-HFD-WT-2025-9-4-17-13.tar.gz](https://github.com/user-attachments/files/22162789/supportdata-WA1USA-CHOPMIST-HFD-WT-2025-9-4-17-13.tar.gz)

[supportdata-WA1USA-CHOPMIST-HFD-WT-2025-9-4-17-28.tar.gz](https://github.com/user-attachments/files/22162794/supportdata-WA1USA-CHOPMIST-HFD-WT-2025-9-4-17-28.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

edkraus, 2025-09-05:
One of the indicators we had problems at a large remote site with multiple Mikrotik devices and a camera feed was inability to get a good camera feed.  We also had meshinfo glitching on us.  3.25.8.0 did not work well, and it took until the Sept 4 nightly to get it working.  At this point we are back up and running with good network speed.  I'd go directly to 0904 or higher with a dedicated path from your camera to you as an initial test.  It takes a while longer than you expect for the new firmware to establish stable routing and start working well.  73  Ed

denis133, 2025-09-06:
Running much better on 904 BUT it changed the channel once again from 149 to 183 on node [supportdata-WA1USA-CHOPMIST-HFD-WT-2025-9-4-17-28.tar.gz](https://github.com/user-attachments/files/22162794/supportdata-WA1USA-CHOPMIST-HFD-WT-2025-9-4-17-28.tar.gz)  
Not a big deal now that I expected it.

closed 2025-09-24 by denis133