aredn/aredn #2429: Static routes in /etc/aredn_include/static_routes do not appear to be getting applied

convolved-KY4G, 2025-09-05:
### Describe the bug
Static routes in /etc/aredn_include/static_routes do not appear to be getting applied upon booting the AREDN node 

- This is occurring in release 3.25.8.0

### Expected behavior
That the routes specified in the /etc/aredn_include/static_routes file would get added and be shown when running "ip route show".  It is possible that I have something wrong in the file, but I really don't think that is the issue.

### Support data

[supportdata-KY4G-10-WINLINK-2025-9-5-17-7.tar.gz](https://github.com/user-attachments/files/22181806/supportdata-KY4G-10-WINLINK-2025-9-5-17-7.tar.gz)

### Screenshots
Contents of /etc/aredn_include/static_routes

[image]

Results of "ip route show" after node boots up and the static route that was expected is not there. Note that I then  tried adding a route via the CLI after the fact and those results are shown. I thought was how you would do it from the CLI and it seemed to take it.

[image]

### Additional context
I'm really trying to get a laptop on the AREDN LAN (br-lan) to be able to access a service from a server on a different  network in my household (basically backup the laptop).  No packets should need to leave my premises to accomplish this.  Getting the static route in the AREDN node is the first step because if it hits the ISP it has no idea where to route it. The AREDN node LAN side, WAN side, and the backup service are all on separate VLANs on my router.  The WANs then get routed to the ISP.  I do realize there is more to getting the traffic onto the other VLAN, but getting the route in there persistently is the first step.  I am not an expert on this.

aanon4, 2025-09-05:
Did you run node-setup after changing the static route file and then rebooting?

convolved-KY4G, 2025-09-05:
No I did not at first.  I didn't see that in the documentation I had looked through.  After you asked, I looked into it

[supportdata-KY4G-10-WINLINK-2025-9-5-17-45.tar.gz](https://github.com/user-attachments/files/22182749/supportdata-KY4G-10-WINLINK-2025-9-5-17-45.tar.gz)

 and did find some information about it.  I rebooted again, ran "node-setup and go the same result.  I then ran "node-setup route_to_nas" and got the same result.  I hope that was the right argument to provide.  I also tried "node-setup /etc/aredn_include/static_routes".  None of that seems to make any difference.  I could not find the string /etc/aredn_include/static_routes in the script, so I'm not sure the file is being processed.  Some of the documentation I found made me think I had to provide an argument, but that may have been a really old 2015 perl file.  I added updated support data in case it was needed.

convolved-KY4G, 2025-09-06:
Thanks for making those additions to the documentation.  Closing this issue.

closed 2025-09-06 by convolved-KY4G