aredn/aredn #2432: hAP ac3 time is UTC whether Time Zone is UTC or Am./Chicago

K5RA, 2025-09-07:
My hAP ac3 running 3.25.8.0 displays UTC whether Time Zone is UTC or American/Chicago.  It is isolated on my bench, with only Ethernet and Win 11 desktop connections.  I have other hAP nodes that behave correctly, so probably not firmware.  How can this happen?? 

[supportdata-K5RA-HAP32-2025-9-7-1-52.tar.gz](https://github.com/user-attachments/files/22192560/supportdata-K5RA-HAP32-2025-9-7-1-52.tar.gz)

[image]

K5RA, 2025-09-07:
Local real time (lower left of the screenshot) is on the Win 11 PC.  I assume it is ntp.  It agrees with real time on my cell phone, so it is CDST (GMT-5).  
I believe the node gets time from ntp.  When I first powered it up, it had no WAN and time was crazy.  Plugged in WAN, selected continuous updates, and the minutes part settled in fairly quickly, and the display indicated ntp.

K5RA, 2025-09-08:
I upgraded the node to latest nightly build.  Node clock still displays GMT when Time Zone is Am/Chicago.  

[image]

[supportdata-K5RA-HAP32-2025-9-8-3-53.tar.gz](https://github.com/user-attachments/files/22200874/supportdata-K5RA-HAP32-2025-9-8-3-53.tar.gz)

aanon4, 2025-09-08:
So what other timezones have you tried?
What is the time if you log into the node and look?
What timezone does the system (not the UI) say it is using?