aredn/aredn #2437: Mikrotik hAP ac2 in reboot cycle after committing a Wireguard Tunnel

Orv, 2025-09-09:
### Describe the bug
Upon committing a Wireguard Tunnel, my localnode (hAP ac2) goes into a reboot cycle, even though it seems to have plenty of RAM

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
A clear and concise description of what you expected to happen.
Not this...

[supportdata-W6BI-SHACK-NODE-2025-9-8-17-46.tar.gz](https://github.com/user-attachments/files/22223937/supportdata-W6BI-SHACK-NODE-2025-9-8-17-46.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-09-09:
Unfortunately nothing in the logs - perhaps unsurprising as after a reboot there is no logging of the reboot event which would be in ram.

aanon4, 2025-09-09:
Reverting the panic change for now.