aredn/aredn #2465: Converted supernode to Babel-only. Now only sees a fraction of tunnels, and can't reach them.

Orv, 2025-09-19:
### Describe the bug
Converted supernode to Babel-only. Now only sees 5 of 11 tunnels, and can't reach the ones it displays.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
Since all of the supernodes I have tunnels to are Babel-enabled, would have expected all the tunnels to be functional.

### Support data

[supportdata-W6BI-SOCAL-SUPERNODE-2025-9-19-21-9.tar.gz](https://github.com/user-attachments/files/22434826/supportdata-W6BI-SOCAL-SUPERNODE-2025-9-19-21-9.tar.gz)

### Screenshots

[image]

### Additional context
Add any other context about the problem here.

aanon4, 2025-09-19:
Please tell me that you either took a backup before this upgrade or at least got support data before the upgrade?

Orv, 2025-09-19:
Of course.

Orv, 2025-09-19:
Most recent backup.  Not really a zip file; had to rename it to attach it to the ticket.

[backup-W6BI-SOCAL-SUPERNODE-2025-9-17-18-50.backup.zip](https://github.com/user-attachments/files/22435560/backup-W6BI-SOCAL-SUPERNODE-2025-9-17-18-50.backup.zip)

Orv, 2025-09-24:
Replaced hardware and backed up to Babel+OLSR for now.