aredn/aredn #2467: Babel routing not working (again)

edkraus, 2025-09-21:
### Describe the bug
Current node status page shows only 20 Babel capable nodes out of several hundred, and very few nodes can be reached by searching via Babel routing.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes

### Expected behavior
A clear and concise description of what you expected to happen.
I would like firmware to see and list all Babel capable nodes ... and not drop most of them over a period of time.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
Attached

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

edkraus, 2025-09-22:
OK, no reply.  The two nodes nearest to me are my node and the one it is tunneled to.

I rebooted my hAP2 and it comes back only 20 Babel nodes.  I rebooted the other end of the tunnel (server) and now it only has 18 Babel nodes.

Something very clearly is wrong, and rebooting is not restarting Babel routing.  This is different than the problems we have had since 3.25.8.0 ... every time restarting restored routing for a while.

Ed

aanon4, 2025-09-22:
Ed - I'm currently unwell so I'm afraid responses will be slow and brief.
Work out where the 20 nodes gets you to and, somewhere at that perimeter I suspect you'll find a node with a crashed babel daemon.

edkraus, 2025-09-25:
Affected sites still were routing ok this afternoon.  I looked again tonight and only 33 Babel nodes.  The bad tunnels are all turned off.  Any node that was previously suspect has been rebooted.  No improvement.