aredn/aredn #2498: Babel only routing changes Traceroute and Ping Tools

edkraus, 2025-09-28:
### Describe the bug
On parts of the network that are behind a Babel only build, the traceroute and ping tools built into the AREDN UI node tools fail to work, even though the devices can be reached via Babel routing in the search tools.  Nodes that are beyond the Babel only choke point fail to show up in the drop down menus.  When I switch from the node tools available on the GUI to the same tools available by command prompt I can reach the distant nodes.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
I would like the UI node tools to find and display all the nodes in the network even if OLSR is not available, as this is the only way to check these functions from remote nodes.  

### Support data

[supportdata-KE7AWR-OHSU-SXT1-2025-9-28-15-3.tar.gz](https://github.com/user-attachments/files/22583452/supportdata-KE7AWR-OHSU-SXT1-2025-9-28-15-3.tar.gz)

aanon4, 2025-09-28:
Works as designed. We use the OLSR list of nodes for the OLSR+Babel node and the Babel list for Babel-only. You can still type any node name into the selection box - you dont have to pick it from the list.

closed 2025-09-28 by aanon4

edkraus, 2025-09-28:
OK, I see that.  We now presume we know the host name of the nodes we might not be able to see ... but once everything is Babel that should be resolved.
Interesting times, and thank you.
Ed Krausf.t.n.m.p.