aredn/aredn #2499: /sbin/sysupgrade leaves multiple user_agent options in /etc/opkg.conf

pmilazzo, 2025-09-28:
### Describe the bug
After performing a manual upgrade using "`/sbin/sysupgrade /tmp/local_firmware`", the `/etc/opkg.conf` file will contain two `user_agent` options: one representing the previous firmware version, and the other representing the upgraded firmware version.

- [x] Does this issue still occur on the latest version? (**Required**) YES

### Expected behavior
Only a single option, representing the upgraded firmware version, should be present in the file.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-K3PGM-NPR-2025-9-28-19-23.tar.gz](https://github.com/user-attachments/files/22584833/supportdata-K3PGM-NPR-2025-9-28-19-23.tar.gz)
[supportdata-W3DRA-88-8-49-2025-9-28-19-25.tar.gz](https://github.com/user-attachments/files/22584834/supportdata-W3DRA-88-8-49-2025-9-28-19-25.tar.gz)

### Screenshots
#### Upgrading from 3.25.8.0 to nightly 20250928-dadf66e3:
```
root@K3PGM-NPR:~# cat /etc/opkg.conf
dest root /
dest ram /tmp
lists_dir ext /var/opkg-lists
option overlay_root /overlay
option check_signature
option user_agent opkg-3.25.8.0
option user_agent opkg-20250928-dadf66e3
root@K3PGM-NPR:~#
```

#### Upgrading to latest TinyBabel version (which is up to date with the upstream "babel" branch as of this morning):
```
root@W3DRA-88-8-49:~# cat /etc/opkg.conf
dest root /
dest ram /tmp
lists_dir ext /var/opkg-lists
option overlay_root /overlay
option check_signature
option user_agent opkg-TinyBabel-babel-75751f06
option user_agent opkg-TinyBabel-babel-e42039cc
root@W3DRA-88-8-49:~#
```

### Additional context
Add any other context about the problem here.

aanon4, 2025-09-28:
We do not support upgrading with /sbin/sysupgrade - please use /usr/local/bin/aredn_sysupgrade. If the problem persists when using that please let me know.

pmilazzo, 2025-09-28:
Argh! I totally missed that change; sorry!

No, the problem does not appear when using `/usr/local/bin/aredn_sysupgrade` to perform the upgrade. Sorry for the confusion!

closed 2025-09-30 by aanon4